Ninoh-FOX/POCKETGO2_ROGUE_CFW #1: Cannonball does not work with 1.0.2 but with original firmware

miwasp, 2020-01-14:
Cannonball (https://github.com/gameblabla/Cannonballs) does not work with version 1.0.2. It just falls back to gmenux without any error. This works with the original firmware.
I compared strace logs from ofw and cfw and at some point where ofw loads /usr/lib/ligGLESv2.so.2 cfw wants to load libGL.so.1,  which does not exist.

Log of ofw:
write(1, "Unable to load widescreen tilema"..., 35Unable to load widescreen tilemaps
) = 35
stat("/etc/ld.so.cache", {st_mode=S_IFREG|0644, st_size=11659, ...}) = 0
open("/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 9
mmap(NULL, 11659, PROT_READ, MAP_SHARED, 9, 0) = 0x777f8000
close(9)                                = 0
open("/usr/lib/libGLESv2.so.2", O_RDONLY) = 9
fstat(9, {st_mode=S_IFREG|0755, st_size=54596, ...}) = 0
close(9)                                = 0
open("/usr/lib/libEGL.so.1", O_RDONLY)  = 9

Log of cfw:
write(1, "Unable to load widescreen tilema"..., 35Unable to load widescreen tilemaps
) = 35
stat("/etc/ld.so.cache", {st_mode=S_IFREG|0644, st_size=14987, ...}) = 0
open("/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 10
mmap(NULL, 14987, PROT_READ, MAP_SHARED, 10, 0) = 0x7718d000
close(10)                               = 0
open("/lib/libGL.so.1", O_RDONLY)       = -1 ENOENT (No such file or directory)
open("/lib/libGL.so.1", O_RDONLY)       = -1 ENOENT (No such file or directory)
open("/usr/lib/libGL.so.1", O_RDONLY)   = -1 ENOENT (No such file or directory)
munmap(0x7718d000, 14987)               = 0
munmap(0x76d18000, 1056768)             = 0
munmap(0x76c86000, 598016)              = 0
munmap(0x76458000, 188416)              = 0
exit_group(0)                           = ?
+++ exited with 0 +++

emulationstation (https://github.com/ManuelSch81/RG350-EmulationStation_configured) also does not work (but works with ofw). I think thats the same problem.

Ninoh-FOX, 2020-01-14:
thanks, I am fixing this

Bodenseematze, 2020-01-15:
I don't know, if this is related - but I have a lot more apps which don't run at the Rogue CFW 1.2 but did at original Fw; e.g. the Arkanoid game which was included in my OFW stick.
At the CFW stick it does not even show up in section "games".
An when I try to directly execute it (e.g. with DinguxCmdr) it directly returns to gmenux.

But I'm able to make it work when I start the "Esoteric"-GUI - there I have to start it with "OPK launcher" - then it works!

miwasp, 2020-01-15:
> I don't know, if this is related - but I have a lot more apps which don't run at the Rogue CFW 1.2 but did at original Fw; e.g. the Arkanoid game which was included in my OFW stick.

I haven't tested the package thats included in the ofc but the one from here (https://github.com/retrogamehandheld/OpenDingux) runs fine.
Just copy the opk file to the second sdcard into the folder "apps" and it shows up in the section games and is playable.

Bodenseematze, 2020-01-15:
You're right - the Arkanoid.opk from that page runs fine!
Don't know, what's the problem with my original .opk was/is...

Ninoh-FOX, 2020-01-18:
Fixed in the last release 1.0.3

closed 2020-01-18 by Ninoh-FOX