Ninoh-FOX/POCKETGO2_ROGUE_CFW #3: R2 button only registers when double-clicked

Andrew-H2O, 2020-01-16:
Not sure if this is a CFW problem or a 'PocketGo test' problem, but in the PocketGo test app, the R2 button only registers if double-clicked.

Also, pressing either of the volume buttons shows both activating onscreen, with the following displayed below;
Last detected key: 0[0x0000] Not defined

magimog, 2020-01-17:
Can confirm the same as Andrew-H20. R2 only registers after several presses, between 2-7 for me. Applies both in PocketGo test app and ingame. Switching to OFW the issue is not present.

Ninoh-FOX, 2020-01-18:
Fixed this in the new release 1.0.3

closed 2020-01-18 by Ninoh-FOX

magimog, 2020-01-18:
This issue looks to be still appearing on 1.0.3 for me, but slightly different.

1. Behavior has decreased from 2-7 presses to register to just two. After it registers I can press it multiple times without issue. Tested with PocketGo Tester app and the R2 prompt to show button has been pressed seems to stick for a very long time. PSX emulator showed similar behavior

2. When Using the Power+B option to match Dpad to Analog, phantom Up presses register in PocketGo tester app. Wondering if this is due to the touchiness of my analog stick registering up. Maybe a deadzone issue?