Ninoh-FOX/POCKETGO2_ROGUE_CFW #4: Sound volume jumps from 32 to 0

miwasp, 2020-01-17:
Sound volume jumps from value 32 directly to value 0 (mute). The value 32 is already really loud, so one can only choose between loud and mute.

Ninoh-FOX, 2020-01-18:
this is a problem of hardware, the one solution is hardmod.

closed 2020-01-18 by Ninoh-FOX

fornellas, 2020-01-20:
@Ninoh-FOX are you aware of any existing hardware mods? I suppose I'd have to solder some resistors in parallel with both the headphone jack and the speaker to dissipate power and lower the sound... I'm surprised it is like this, it is too bad.

I had a look [at the drivers](https://github.com/Ninoh-FOX/POCKETGO2_ROGUE_CFW/tree/20df64e660f04cf56f19234aec5b61a7ea37b5bc/kernel/sound/soc/jz4770) for the sound chip, didn't spot anything related to volume there. Do we ask the hardware "raise the volume" and the chip reports back "yep, went from 0 to 32", is that it?

Ninoh-FOX, 2020-01-22:
tomorrow I go to do the hardmod and update the steps.

Ninoh-FOX, 2020-01-22:
and the resistors is in serial in the positive wire, not in parallel.

fornellas, 2020-01-22:
https://photos.app.goo.gl/78RZJYCTSH4btwBi6

Here's how I managed to fix mine for reference. I used a small pot for volume control.

Headphone jack is still quite loud though, I was wondering if there's some gain control register on the sound chip, but I could not find a datasheet for it.