Ninoh-FOX/POCKETGO2_ROGUE_CFW #12: Version 1.0.3 - R2 not working properly

ecsj88, 2020-01-20:
I have tried on some games and on the PocketGo2 test app. It works for some time, then it stops working after sometime.

magimog, 2020-01-21:
Placing my comment from closed issue #3 here:

This issue looks to be still appearing on 1.0.3 for me, but slightly different.

Behavior has decreased from 2-7 presses to register to just two. After it registers I can press it multiple times without issue. Tested with PocketGo Tester app and the R2 prompt to show button has been pressed seems to stick for a very long time. PSX emulator showed similar behavior

When Using the Power+B option to match Dpad to Analog, phantom Up presses register in PocketGo tester app. Wondering if this is due to the touchiness of my analog stick registering up. Maybe a deadzone issue?

Ninoh-FOX, 2020-01-25:
In my console (GCWZERO_V1.2 2019.11.26 board) everything works correctly, I don't know if there will have been revisions of them, but not having the original code I don't know where the fault can be.

magimog, 2020-01-26:
Opened up my unit and it's the same board revision as yours. Had to revert to stock firmware to alleviate this issue. 
Is there anything that I can provide from stock firmware that can help troubleshoot?

Ninoh-FOX, 2020-01-29:
fixed in the new release

closed 2020-01-29 by Ninoh-FOX