Ninoh-FOX/POCKETGO2_ROGUE_CFW #13: Screen flashing issue in CFW 1.0.3

Andrew-H2O, 2020-01-20:
If the second SD card has a folder called 'apps', containing the same apps as on the card in slot 1, the launcher will just flash the background screen with no icons.  The only way to break out of this is to eject the card in slot 2, and/or rename the folder to something other than 'apps'

Not sure if this also happens in previous revisions, but I noticed it when trying to restore apps and roms after a fresh install of CFW 1.0.3

It doesn't happen with the FW that ships with the PG2
It also doesn't happen with only 1-2 .opk files in the apps folder in card 2, but it does with ~40

magimog, 2020-01-21:
Placing my comment from closed issue #3 here:

This issue looks to be still appearing on 1.0.3 for me, but slightly different.

Behavior has decreased from 2-7 presses to register to just two. After it registers I can press it multiple times without issue. Tested with PocketGo Tester app and the R2 prompt to show button has been pressed seems to stick for a very long time. PSX emulator showed similar behavior

When Using the Power+B option to match Dpad to Analog, phantom Up presses register in PocketGo tester app. Wondering if this is due to the touchiness of my analog stick registering up. Maybe a deadzone issue?

Ninoh-FOX, 2020-01-25:
gmenu2x may conflict with having the same opk on both routes, anyway. Why do you have repeated applications? Even so, depending on the format of the sd2, the applications folder has to be APPS in fat32 and apps in ext3 / 4

Andrew-H2O, 2020-01-26:
I had the apps folder on SD 2 because I was restoring them to SD 1 after re-flash of the card.  I copied across from SD 2 to SD 1, then exited DinguxCmdr, at which time the screen flashed, and the only way to stop it was to eject SD 2, and rename the apps folder on it to something else (like apps-backup)

On the good side, it seems that the apps could all be kept on SD 2, which would make re-flash of SD 1 even quicker to reconfig after.

Ninoh-FOX, 2020-05-27:
Fixed in the last release

closed 2020-05-27 by Ninoh-FOX