Ninoh-FOX/POCKETGO2_ROGUE_CFW #14: DinguxCmdr file rename issue

Andrew-H2O, 2020-01-20:
The footer below the virtual keyboard used for renaming files in DinguxCmdr incorrectly lists X as backspace and Y as space.  The keys are actually reversed.

It looks to be connected to lines 78-87 in keyboard.cpp in the source, where it looks like it's been compiled for the GCW0

`    // Create footer
    m_footer = SDL_utils::createImage(SCREEN_WIDTH, H_FOOTER, SDL_MapRGB(Globals::g_screen->format, COLOR_BORDER));
#if defined(PLATFORM_DINGOO)
    SDL_utils::applyText(SCREEN_WIDTH >> 1, 1, m_footer, m_font, "A-Input   B-Cancel   START-OK   L/R-Change   Y-Backspace   X-Space", Globals::g_colorTextTitle, SDL_utils::T_TEXT_ALIGN_CENTER);
#elif defined(PLATFORM_GCW0)
    SDL_utils::applyText(SCREEN_WIDTH >> 1, 1, m_footer, m_font, "A-Input   B-Cancel   START-OK   L/R-Change   X-Backspace   Y-Space", Globals::g_colorTextTitle, SDL_utils::T_TEXT_ALIGN_CENTER);
#else
    SDL_utils::applyText(SCREEN_WIDTH >> 1, 1, m_footer, m_font, "Ret-Input  Bsp-Cancel  w-OK  PgU/PgD-Change  q-Backspace  a-Space ", Globals::g_colorTextTitle, SDL_utils::T_TEXT_ALIGN_CENTER);
#endif`

Not sure if this is in-scope, but thought I'd mention it.

Ninoh-FOX, 2020-01-21:
Ok, thanks

Ninoh-FOX, 2020-01-22:
yes, this is for gcwzero, I go to change this soon

Ninoh-FOX, 2020-01-26:
Fixed for the next release

closed 2020-01-26 by Ninoh-FOX