Ninoh-FOX/POCKETGO2_ROGUE_CFW #8: Updated RootFS is corrupted!

pasel, 2020-01-20:
Hi,
when i run the update on my device, after the the verifying of the checksum I get the following message:

"ERROR!
Updated RootFS is corrupted!"

How can I solve this?

Thanks!
Pascal

Ninoh-FOX, 2020-01-20:
I am flashing imagen_sd.bin in a new sdcard and late I am update for opk and all work fine.

Do you have a good quality sdcard or if the sdcard that arrive with the console?

pasel, 2020-01-20:
Hi, I use a Transcend 32Gb Premium SD Card

Ninoh-FOX, 2020-01-20:
Do you extended the ext4 partition from pc when you flashed the sdcard image?

pasel, 2020-01-20:
no

Ninoh-FOX, 2020-01-20:
Then I don't know why not work for you.

Andrew-H2O, 2020-01-20:
I had the same happen with a 32GB Toshiba Exceria card.  I ran the update.opk and at the end got the same message.  It rebooted back to 1.0.2  
I ended up reflashing the card and then reimporting all of the emulators and roms.

zaper4life, 2020-01-20:
I get same issue with all versions and stuck on black screen after setup complete 
(if via opk I get the "ERROR! Updated RootFS is corrupted!" message)

Ninoh-FOX, 2020-01-21:
Use Etcher or dd command if win32dw not work

miwasp, 2020-01-21:
I am flashing sd_image.bin to a new sdcard (32GB Sandisk) and let the pocketgo extend the partition and filesystem. Then i start the update opk and get this "rootfs corrupted" message. It makes no difference if i write the image with Win32DiskImager, Etcher or dd.
Does it make any difference if the update opk is started from the first or second sdcard? If started from the second sdcard, does the used filesystem (fat32, exfat, ntfs, ext4) make any difference?

Ninoh-FOX, 2020-01-21:
No, this doesn't have difference for file type, tomorrow I go to reupload the update opk.

Ninoh-FOX, 2020-01-21:
Update opk only work from gmenu2x, not for esoteric

miwasp, 2020-01-30:
I never used esoteric.
Same error again with update 1.0.4.
I did some more tests. If i copy the contents of the opk-file into a subdir on the second sdcard, connect via ssh and start update.sh from there everything works and the update is done.

If i just start the opk out of gmenu2x as usual, i get the error "rootfs corrupted". Then i started the opk out of gmenu2x and let it stay at the first dialog and connected via ssh:
```
POCKETGO2:/media/data/local/home # sha1sum /mnt/OS\ Update\ 1.0.4/rootfs.squashfs
cfa6e98f435614a6c798c4ccff60ebb1a726a549  /mnt/OS Update 1.0.4/rootfs.squashfs
```
This was just to verify that the sha1sum of the source is correct. Same as in the sha-file->OK
Then i copied the new rootfs manually to the temporary mount, synced and tested the sha1sum:
```
POCKETGO2:/media/data/local/home # cp /mnt/OS\ Update\ 1.0.4/rootfs.squashfs /mnt/_system_update/update_rootfs.bin
POCKETGO2:/media/data/local/home # sync
POCKETGO2:/media/data/local/home # echo 3 > /proc/sys/vm/drop_caches
POCKETGO2:/media/data/local/home # sha1sum /mnt/_system_update/update_rootfs.bin
d24588e30c2af1ba4f90b3852ca004cda69fdb15  /mnt/_system_update/update_rootfs.bin
```
I tested this several times and also tried to copy the new rootfs to the second sdcard, the sha1sum was never the same!?
If i check the sha1sum of the source again after copying i get an error:
```
POCKETGO2:/media/data/local/home # sha1sum /mnt/OS\ Update\ 1.0.4/rootfs.squashfs
sha1sum: can't read '/mnt/OS Update 1.0.4/rootfs.squashfs': Input/output error
```
Something must be seriously wrong with the temporarily mounted squashfs from the opk.

Ninoh-FOX, 2020-05-27:
Erase the sha1 files and try again.

closed 2020-05-27 by Ninoh-FOX