Ninoh-FOX/POCKETGO2_ROGUE_CFW #15: Mouse doesn't work in DOSBox

maksym-kershengolts, 2020-01-21:
I tried to play a point-and-click quest in DOSBox (Legend of Kyrandia), and I was only able to skip the intro by pressing Y, I guess it acts like Esc or Enter.
However the further game is supposed to be controlled by mouse, but no controls seem to move mouse pointer or click mouse buttons.
dosbox.conf file looks like a standard one from PC version, and doesn't have any options to bind console's stick and buttons to control mouse.
Any suggestions?

Ninoh-FOX, 2020-01-21:
the mouse work fine to me, for that the mouse emulator work fine, first you need shutdown the screen in gmenu (I don't know but work) with the button power, then turnon again the screen with button power and then, in dosbox, press power+r1 for activate the mouse emulator, stick move the mouse and L2 is left-button and R2 is right-button.

MaxMarauder, 2020-01-21:
@Ninoh-FOX I have a weird issue with the mouse emulator - the cursor constantly moves to the left. When I pull the stick right, it just moves a little bit and then sticks to the left edge once again.
The same happens in the PocketGo2 test app.
But in the normal mode the stick works fine, so it doesn't seem to be a hardware issue.

Andrew-H2O, 2020-01-21:
@Ninoh-FOX I have the exact same problem as @MaxMarauder.

Ninoh-FOX, 2020-01-22:
yes yes, I know, for that the mouse emulator work fine, you need follow the steps that I said (original fw don't have mouse emulator), anyway I am tried fix this

closed 2021-02-16 by Ninoh-FOX