Ninoh-FOX/POCKETGO2_ROGUE_CFW #17: Rogue 1.0.3 - Laggy gaming as soon as I turn it on

ecsj88, 2020-01-22:
I have noticed that from time to time games lag for a couple of minutes as soon as I turn the PG2 on if I hadn't been playing with it for a couple of hours. Sometimes I gotta restart it twice to make games run smoothly. Is this a common problem?

Ninoh-FOX, 2020-01-25:
I haven't noticed any of that yet, let's see if I play a little and confirm it. In which game, emulator, have you noticed them?

ecsj88, 2020-01-28:
Suddenly everything is okay for the past 5 days! It was acting like that for Street Fighter 3: Third Strike and Three Wonders, both for FBA. Megaman X for SNES and Super Mario Bros 2 For NES. I have tested them all again and no problem.

closed 2020-01-28 by ecsj88