Ninoh-FOX/POCKETGO2_ROGUE_CFW #18: Restore alsa settings on boot

fornellas, 2020-01-22:
Related to #4 , we can mitigate the headphone high volume (speaker still needs a hardware mod).

Alsa has a channel for Headphone and another channel for PCM. The latter is mapped to the physical buttons. The Headphone channel is applied over the PCM channel. It boots on 70, but lowering it to 3, let's the physical buttons volume control work a lot better for headphones.

Can we have:

- On power down, run `alsactl store -f alsa.settings`.
- On boot run `alsactl restore -f alsa.settings`.

I'd recommend also shipping the firmware with an `alsa.settings` with the Headphones channels lowered to 3 (not the default 70).

Fair?

Ninoh-FOX, 2020-01-25:
So you want the headphones to always start at level 3 instead of level 70? I think you can configure that in alsamixer anyway.

fornellas, 2020-01-25:
It can be configured, but the configuration is lost after power off / power
on, my suggestion is to save / restore it.

fornellas, 2020-02-10:
https://github.com/OpenDingux/alsa-volume
Just found this.

Ninoh-FOX, 2020-05-08:
I think that I can do this in the next release

Ninoh-FOX, 2020-05-10:
fixed, I go to close the theme.

closed 2020-05-10 by Ninoh-FOX