Ninoh-FOX/POCKETGO2_ROGUE_CFW #21: Switch on backlight after starting

johnhugopete, 2020-01-26:
It is possible to switch off backlight completely and nothing is visible on the screen. I think it does not have any sense to keep this after restart - in that case nothing is visible. Of course, it can be fixed by the button shortcut, but I have heard about some people thinking they have broken screen when this happened for the first PocketGo. Here for the new PocketGo it is possible as well - it would be good if the firmware checked that the backlight is off when starting and set there some minimum.
I see no real use case in starting the PocketGo with the backlight off - since it is completely unusable.

Ninoh-FOX, 2020-01-26:
I do not understand anything you want to say, the console already turns off the screen in gmenu with the power button.

johnhugopete, 2020-01-26:
Well, you can set brightness so low that nothing is visible by Power + Volume down. And I think this is saved... So the next time the console is restarted, it will end up with the black screen. Of course you can increase brightness, but I remember some people were thinking their screen went bad with the first PocketGo. I think here you will get visible splash screen, but then the firmware applies the stored brightness  and there is black screen. I guess the firmware should allways set non-zero brightness after start.

Ninoh-FOX, 2020-01-26:
Who will lower the screen brightness to 0 and reboot?  Has no sense.

johnhugopete, 2020-01-26:
> Who will lower the screen brightness to 0 
> and reboot? Has no sense.

You are right. :-) But if you look through the PocketGo forums, you find cases of people complaining that that their PocketGo is broken only being told that they should try to increase brightness and then thanking it was the case (I definitely saw this in discusions concerning the old PocketGo, but I think it can happen here as well).

Well, not that important, just thought it a nice to have. I guess probably not worth to spend long discussing - so feel free to close.

Ninoh-FOX, 2020-01-26:
Ok

closed 2020-01-26 by Ninoh-FOX