Ninoh-FOX/POCKETGO2_ROGUE_CFW #22: Analog

NexD2, 2020-01-27:
my Analog stopped working.  he stayed with Drift, there's no way to use the console anymore.  Is there no way to permanently disable the analog?  not just on the menus, because I know the new CFW did that.  but it would be nice if there was a way to disable it in general, in emulators.

suicicoo, 2020-05-06:
i just joined to suggest the same - it would be awesome if you could disable the analog input!

Ninoh-FOX, 2020-05-06:
I am make this possible in the next release (only ones hours)
https://github.com/Ninoh-FOX/RG350_pwswd/commit/2a224f1cf344cc5cdd3c0e90e2002589660ee109

Ninoh-FOX, 2020-05-06:
update with the new key combo for disable de analog stick (POWER+L1), closed.

closed 2020-05-06 by Ninoh-FOX