Ninoh-FOX/POCKETGO2_ROGUE_CFW #24: Clock time doesn't save when powering off

Johnny56732, 2020-01-31:
The time is always wrong, and when I adjust for the correct time save it & exit then power off and turn on again the click shows the wrong time and didn't save the changes that I made
It's currently 5:59pm in the U.S. where I'm at and the pocket go shows 
18:54  military time (24hr clock)
But if you click on the clock app it shows
6:59 PM (12hr clock)
No matter how many times I set it it is 1 hour off and will only display military time unless you click on to the click app which will show a 12 hour clock
Please fix

Ninoh-FOX, 2020-02-01:
do you use esoteric or gmenu2x?

Johnny56732, 2020-02-01:
Gmenu2x

fornellas, 2020-02-10:
I don't recall the hardware having a RTC...

Ninoh-FOX, 2020-02-13:
The hardware have RTC

Ninoh-FOX, 2020-05-06:
the clock work fine in the last releases, closed

closed 2020-05-06 by Ninoh-FOX