Ninoh-FOX/POCKETGO2_ROGUE_CFW #25: Oswan emulator playing at 76fps

billbop, 2020-02-06:
Hi! First time writing! First off this pocket go is awesome, and this cfw is awesome too! 

So to the issue: the oswan emulator plays both wonderswan and wonderswan color games at 76fps, with no option to lock at 60. Any help would be great!

billbop, 2020-05-03:
Anyone else have this issue? If not, would you be willing to share oswan opk?

Rican7, 2020-05-24:
@billbop I came here after seeing [your post on another thread](https://github.com/Ninoh-FOX/POCKETGO2_ROGUE_CFW/issues/35#issuecomment-626643886) (which, by the way, was off-topic, rude, and incredibly self-entitled).

The Oswan emulator isn't managed by this code-base. You'll have to report/talk to the creators/contributors of that emulator:

https://github.com/gameblabla/oswan _(I'm pretty sure this is it... I'm not positive)_