Ninoh-FOX/POCKETGO2_ROGUE_CFW #29: How to fix PCM (Speaker) loud volume (edit asound.conf)

Shut3n, 2020-04-14:
Hi,

I managed to enable "Master" volume control on the alsa-mixer by editing /etc/asound.conf (/etc on the rootfs.squashfs file)

==ROGUE ORIGINAL==
defaults.ctl.card 1;
defaults.pcm.card 1;

==REPLACE WITH THIS LINES==
pcm.!default  {
 type plug     
 slave.pcm "speakerphat"
}

ctl.!default {
 type hw card 1
}

pcm.speakerphat {
  type softvol
  slave.pcm "plughw:1"
  control.name "Master"
  control.card 1
}

I had to reboot the handheld, open "Sound Mixer", close it and open again (first opening didn't appeared). It enables a new virtual controller that works as "Master" one, so it is fixable the PCM loud volume without any wiring or soldering necessary. 

Please check that is working for everyone.
It worked for me on a Playgo with Rogue 1.0.4

link125, 2020-04-15:
How did you edit /etc/asound.conf?

idonda, 2020-04-15:
is this a fix for loud volume at the minimum level (after mute)?

Shut3n, 2020-04-15:
> How did you edit /etc/asound.conf?

I mounted rootfs.squashfs file (in the system SD card, sdcard1), extracted (copy) /etc dir, edit that file and finally merged to rootfs.squashfs file to change it. I did it from linux that recognize both partitions on sdcard, but I'll only need to access to the first partition.

mount rootfs.squashfs /mnt (filesystem is mounted as readonly)
cp -a /etc /home/user/destinaetiondir
gedit /home/user/destinationdir/etc/asound.conf  (replace for the text above and save)

== we create a newone that merges /mnt and edited etc file
mksquashfs /mnt newroot.squashfs -wildcards -e /home/user/destinationdir/etc
mksquashfs /home/user/destinationdir/etc  newroot.squashfs -keep-as-directory
umount /mnt (umount the squashfs)

cp newroot.squashfs rootfs.squashfs (an replace in the first partition of the sdcard1) -- remember make a backup of the original

Shut3n, 2020-04-15:
> is this a fix for loud volume at the minimum level (after mute)?

Yes, with it you enable a new control to adjust Master volume. For me it worked on a Playgo, but I still need a way to it for be saved after poweroff.

link125, 2020-04-15:
Can confirm the changes @Shut3n made do work. Yes after powering off the device, you do have to go into an emulator, then change the Master levels in the "Sound Mixer" app again.

Here are the commands on Linux I used to make the changes to /etc/asound.conf, refining the instructions @Shut3n gave

**First make a backup of rootfs.squashfs, in case something goes wrong you can copy it back in the partition**

Run these commands in the partition directory with rootfs.squashfs

mount rootfs.squashfs /mnt
cp -a /mnt /home/user/pgo (copy the whole /mnt directory, not just /mnt/etc)
gedit /home/user/pgo/etc/asound.conf (replace with the text @Shut3n gave, and then save)

mksquashfs /mnt newroot.squashfs -wildcards -e etc (this is the proper way to exclude the etc directory from our new squashfs. We will add our new /etc with the changed asound.conf in the next command) 
mksquashfs /home/user/pgo/etc newroot.squashfs -keep-as-directory
umount /mnt

mv newroot.squashfs rootfs.squashfs (replacing current rootfs.squashfs with our new one)

So yes, making the speaker quieter is possible without having to do any hardware mods or putting sticky tape over the cover. Although sticky tape is still the easiest option for now!

Shut3n, 2020-04-15:
I've searching fort a way of saving the settings for restoring on the reboot.
The I generate this asound.state file with "alsa store -f  /media/home/asound.state"

```
state.VirMIDI {
	control {
	}
}
state.GCW0 {
	control.1 {
		iface MIXER
		name 'Mic Volume'
		value 0
		comment {
			access 'read write'
			type INTEGER
			count 1
			range '0 - 5'
			dbmin 0
			dbmax 2000
			dbvalue.0 0
		}
	}
	control.2 {
		iface MIXER
		name 'Headphone Playback Volume'
		value.0 20
		value.1 20
		comment {
			access 'read write'
			type INTEGER
			count 2
			range '0 - 31'
			dbmin -2500
			dbmax 600
			dbvalue.0 -500
			dbvalue.1 -500
		}
	}
	control.3 {
		iface MIXER
		name 'PCM Capture Volume'
		value.0 0
		value.1 0
		comment {
			access 'read write'
			type INTEGER
			count 2
			range '0 - 23'
			dbmin 0
			dbmax 2300
			dbvalue.0 0
			dbvalue.1 0
		}
	}
	control.4 {
		iface MIXER
		name 'Line In Bypass Volume'
		value.0 20
		value.1 20
		comment {
			access 'read write'
			type INTEGER
			count 2
			range '0 - 31'
			dbmin -2500
			dbmax 600
			dbvalue.0 -500
			dbvalue.1 -500
		}
	}
	control.5 {
		iface MIXER
		name 'Headphone Switch'
		value true
		comment {
			access 'read write'
			type BOOLEAN
			count 1
		}
	}
	control.6 {
		iface MIXER
		name 'Speakers Switch'
		value true
		comment {
			access 'read write'
			type BOOLEAN
			count 1
		}
	}
	control.7 {
		iface MIXER
		name 'Line Out Source'
		value PCM
		comment {
			access 'read write'
			type ENUMERATED
			count 1
			item.0 PCM
			item.1 'Line In'
			item.2 Mic
		}
	}
	control.8 {
		iface MIXER
		name 'Headphone Source'
		value PCM
		comment {
			access 'read write'
			type ENUMERATED
			count 1
			item.0 PCM
			item.1 'Line In'
			item.2 Mic
		}
	}
	control.9 {
		iface MIXER
		name 'Capture Source'
		value 'Line In'
		comment {
			access 'read write'
			type ENUMERATED
			count 1
			item.0 'Line In'
			item.1 Mic
		}
	}
	control.10 {
		iface MIXER
		name 'PCM Playback Volume'
		value 1
		comment {
			access 'read write'
			type INTEGER
			count 1
			range '0 - 31'
			dbmin -9999999
			dbmax 0
			dbvalue.0 -3000
		}
	}
	control.11 {
		iface MIXER
		name Master
		value.0 129
		value.1 129
		comment {
			access 'read write user'
			type INTEGER
			count 2
			range '0 - 255'
			tlv '0000000100000008ffffec1400000014'
			dbmin -5100
			dbmax 0
			dbvalue.0 -2520
			dbvalue.1 -2520
		}
	}
}

```
With this way is not necessary to do the squash-mount-thing, you can easily copy to the homepart (in the sd card) to access later from the handheld.

Then I can restore the configuration everytime from the terminal app with command:
"alsactl restore -f /media/home/asound.state"

But still not possible to restore from booting. I disabled the /etc/init.d/S90volume.sh script because I intend to save the asound.state to /var/lib/alsa directory.. so it can restore everytime when rebooting.

When I run "alsactl store" I get permission error, not writable , and not found.

Shut3n, 2020-04-16:
Done. It stores the state now.
The passes I follow:
1) Make the squash-mount-thing, then replace etc/asound.conf, comment etc/init.d/S90Volume.sh, and add a new script on etc/init.d/ (can be alsaconfig.sh, I'll copy one example) with alsactl restore|store process.

Then save with the mksquashfs command.

2) Then save an .state (copy the content of the state I generated) to a file like /media/home/asound.state

3) Place the edited rootfs.squash in the card (with new /etc) and the state to home directory). Then you can start the handheld

Shut3n, 2020-04-16:
[image]

I share my edited "rootfs.squashfs" file (with audio config and restoring enabled) and "asound.state" one (to place on /media/home). You can test if it works for yours

[https://drive.google.com/drive/folders/1BHdTk2amB8MBvit-xeKI_ujERpkW2suP?usp=sharing](url)

Ninoh-FOX, 2020-04-16:
Great!! I need test this, I am very very busy in my work with the covid-19 problems, I am everytime out of home cleaning zones biohazard in my country.

sdabrowski, 2020-05-01:
Fixing this issue would correct probably the BIGGEST bug on the Pocket Go 2. It doesn't look like it was corrected in the latest release, was it, @Ninoh-FOX ? I didn't see it listed as a change. 

I hope you can get around to implementing the fix for this issue. I'd do it myself and submit a pull request, but I have unfamiliar with Linux programming. Thanks for all the work you have already done!

Ninoh-FOX, 2020-05-01:
The fix not is perfect because not work with fisica keys and you need run first a opk and late go to sound setting, is for this that not is release this for the moment.

Ninoh-FOX, 2020-05-06:
I finally find that PWSWD can asigned the control volumen to the keys, but only work PCM, I can't change to MASTES:
this is the code: https://github.com/Ninoh-FOX/RG350_pwswd

Ninoh-FOX, 2020-05-09:
i find one mode too of save the volumens after of change any valor, I am release this. theme close.
psy congroo

closed 2020-05-09 by Ninoh-FOX