Ninoh-FOX/POCKETGO2_ROGUE_CFW #33: Default launcher change won't work on simplemenu

phekda, 2020-05-07:
The option Default launcher in simplemenu 5.6.1 doesn't work on the pocket go 2. When I choose YES it will automatically go back to NO after shutdown or reboot. It always fall back to gmenu2x. I am using the latest Rogue FW 1.0.5 on PG2. Apparently it works on RG350 but not on PG2... how is this possible?

phekda, 2020-05-07:
Same effect on the new Rogue 1.0.6 FW

Ninoh-FOX, 2020-05-07:
You may send the question to the SimpleMenu developer

phekda, 2020-05-07:
I already did that but as it seems to work on rg350 i thought maybe it is because of an issue with the firmware as the rg350 and pg2 are based on the same hardware.

Ninoh-FOX, 2020-05-08:
95% the same, pg2 has less libs that rg350 because has less funtions.

phekda, 2020-05-09:
I will close this issue as it seems probably something for the simplemenu developer.

closed 2020-05-09 by phekda