Ninoh-FOX/POCKETGO2_ROGUE_CFW #35: Unable to flash PocketGo v2 (april 2020 edition)

khobar, 2020-05-09:
I just recived PocketGo v2  ( upgraded version from 2020April ) and tried to flash Rogue CFW.
Took brand new 32 GB SD card, formated 2 times with SD Formater in FAT32 .
Then flashed 1.0.6 sd_image_pg2.bin with win32disk imager (tried Etcher as well) - no issues with writing to sd
Then turned off device, put new SD card into TF1 slot , press power for 2-3 sec and nothing happens ( if I put ear to device when pressing power i can hear slight beep ) . Screen is blank, no logo , nothing happens. If i take out batter, put back in the original sd card , device works.
I also tried to flash 1.0.5 and 1.0.0 versions but result is same.

I made img of original sd card, wrote it to the new sd card and the original os works that way, so at least in case of failure I have backup, but I would realy would like to have Rogue CFW installed instead.

Are there any additional steps, something missing on my side ?
Maybe I can do filecopy of Rogue files onto the shipped in OS?

Ninoh-FOX, 2020-05-09:
I do not know what changes will have been made to the new board, but I do not have v2, if not v1, so I do not have to be able to test to get the correct firmware for that version.

Ninoh-FOX, 2020-05-09:
you can try the firmware for RG350, if this works, then I only need change the keys mappers in the kernel.

Ninoh-FOX, 2020-05-09:
said that works or not works then, please.

Midas2310, 2020-05-09:
hey i runn into the same issue got my poket gov2 this week i also got this bussing sound and cant get rid of it tryed using someting from other github to chance the screen hz but it wont load it from the file any ideas ?

Midas2310, 2020-05-09:
ok i tryed the original fm dosent work but the rg350 works and boots fine

khobar, 2020-05-09:
Hi, sorry was away could not reply right away.
I can confirm as well , that SD with latest rg350 firmware properly shows logo starts creating swap etc. and boots fine

Ninoh-FOX, 2020-05-10:
ok, perfect, I need the test messages from the lost buttoms in RG350-test

Ninoh-FOX, 2020-05-10:
ah, and if works the OTG and the Wifi

khobar, 2020-05-10:
Awesome waiting then for some info. 
If needed I can try to provide you with some logs, files or something..

Ninoh-FOX, 2020-05-10:
I am make a test imagen file

Ninoh-FOX, 2020-05-10:
[https://mega.nz/file/6pBVWQqB#v9qaDRNUxAEaecrP5m2sA3Ik9GNhkPa8oca3BxjPiNU](url)

khobar, 2020-05-10:
Flashed, and device booted, however there are some issues with controls
Works: 
D-pad, menu, select, start, ABXY buttons
Not working properly
1. Analog - moving to left is detected as down. right, detected as up, , up detected as right, and down as left
2. Button L2 - 9 [0009] SD:L_TAB is detected as L1]
3. Button L1 is not detected as all
4. Button R2 is not detected
5. Volume buttons not working

Ninoh-FOX, 2020-05-10:
And with the RG350?

Ninoh-FOX, 2020-05-10:
Said me all button id in the two systems please.

khobar, 2020-05-10:
I was using PocketGo2Test , and it was not showing "last key pressed" for mentioned keys.

**1.0.7 PocketGo2 v2 Test image CFW**
D-PAD:
Left : SDLK_LEFT 276 [0x0114]
Right: SDLK_RIGHT 275 [0x0113]
Up: SDLK_UP 273 [0x0111]
Down: SDLK_DOWN 274 [0x0112]

X :  SDLK_SPACE 32[0x0020]
Y :  SDLK_LSHIFT 304[0x00130]
B : SDLK_LALT 308[0x0134]
A : SDLK_LCTRL 306[0x0132]

MENU: : SDLK_RCTRL 305[0x0131]
SELECT : SDLK_ESCAPE 27[0x001B]
START: SDLK_RETURN 13[0x000D]

R1: SDLK_BACKSPACE 8[0x0008]
R2: ?
VOL + : ?
VOL - : ?
L1:?
L2: SDLK_TAB 9[0x0009] (detected as L1)

Also noticed that gui is kind'of blury in GUI (low res?)  , have not tested for emulators

**RG3500 CFW 1.7.9.9**

D-PAD:
Left : SDLK_LEFT 276 [0x0114]
Right: SDLK_RIGHT 275 [0x0113]
Up: SDLK_UP 273 [0x0111]
Down: SDLK_DOWN 274 [0x0112]

X :  SDLK_SPACE 32[0x0020]
Y :  SDLK_LSHIFT 304[0x00130]
B : SDLK_LALT 308[0x0134]
A : SDLK_LCTRL 306[0x0132]

MENU: : ?
SELECT : SDLK_ESCAPE 27[0x001B]
START: SDLK_RETURN 13[0x000D]

R1: SDLK_BACKSPACE 8[0x0008]
R2: SDLK_PAGEDOWN 281[0x0119]
VOL + : 0[0x0000] not defined ( but works ) 
VOL - : 0[0x0000] not defined ( but works ) 
L1: SDLK_TAB 9[0x0009]
L2: SDLK_PAGEUP 280[0x0118]

GUI sharp , compared to 1.0.7 PocketGo2 v2 Test image CFW

Ninoh-FOX, 2020-05-10:
Perfect, them is the same that RG350, the stick work fine in RG350 version.

khobar, 2020-05-10:
Yes , analogue is correctly registering directions ( up is up , down is down etc. )

Ninoh-FOX, 2020-05-10:
Ok, I am try make another imagen in ones days.

Joga29, 2020-05-11:
Bonjour j'ai le même problème impossible d'installer rogue sur la dernière version de la V2 j'ai voulu l'installer sur la carte sd principale en ayant au préalable sauvegarder l'image du FW d'origine mais quand je le remet la console ne s'allume pas non plus auriez vous une solution pour retrouver le FW d'origine ? En plus d'installer le FW pour la rg350 sur une 2e carte SD . Merci d'avance

Ninoh-FOX, 2020-05-11:
Hace tiempo que no hablo frances XD
the cfw for now not work in v2, this week I go try release an update.

Joga29, 2020-05-11:
Oh ok good . For waiting Can I install the rg350 FW ? It work ?

Ninoh-FOX, 2020-05-11:
> Yes , analogue is correctly registering directions ( up is up , down is down etc. ) in RG350

Can you test too if tha new board has rumbe? And too (if you have an OTG usb-c to usb) if this read for example a mouse? (Test all this in RG350 tester app) thanks

Joga29, 2020-05-11:
Ok tell me if I understand correctly , I install the rg350 fw on my Pocketgo V2 ? And After I try to do the tests do you speak

Ninoh-FOX, 2020-05-11:
Yes, but not is his fw, I need dates for that I can do the correct.

Joga29, 2020-05-11:
Ok my Friend  I try it All tomorow if I have thé time and I return to you if it works . So I write the Fw on a formated card in Fat32 and I insert that in the Pocketgo V2 , IS good ?

khobar, 2020-05-11:
Hi,  rumble works in test app in Rg3600 ( when pressed L2+R2)
Unfortunatelly I don't have USB-C OTG to test if mouse works :/

Ninoh-FOX, 2020-05-11:
Perfect, I don't think that otg work because need another port. I now know the changes that I need make in the cfw. Thanks.

Joga29, 2020-05-11:
Ok guys Can you explain me step by step how I install correctly the FW  on the SD card please ? I have write the SD.image.img on an empty card but it doesnt work , the Pocketgo V2 doesnt turn on

khobar, 2020-05-11:
@Joga29 Please check instalation instructions
https://github.com/Ninoh-FOX/POCKETGO2_ROGUE_CFW#installation
You can also find lots of guides on youtube, like this : 
https://www.youtube.com/watch?v=t1fBR0YjJwU

But preferably wait for new cfw from Ninoh-FOX

Joga29, 2020-05-11:
Ok thanks you

billbop, 2020-05-11:
Man, what’s up with this group being prejudice?!?! I opened an issue 3 months ago about the Oswan opk running at 76fps where everywhere else it runs fine. It’s a PGv2 issue only. everyone else’s issues were resolved ahead of me, but mine wasn’t?? 

Can’t someone be cool enough to address why my issue is not being resolved?!! Instead of completely ignore me?!? 

And just to make sure, there’s barely any changes other than problems since the 1.02 update. I would only use that version until there’s a significant upgrade. “If it ain’t broke, don’t fix it.”

And I’m only commenting here cause my issue which is a big one, seems to be completely ignored....

khobar, 2020-05-11:
@billbop Please do not litter, 
Your problem is completely not related to this issue which is unable to flash whole os into newer versions of  PocketGo2, 
Please stick to original one you posted and/or contact 'oswan' maintainers in the first place if you are the only one with that problem

billbop, 2020-05-11:
I’m not littering, as a matter of fact I posted on there 8 days ago, asking for help. But apparently you have to be in some secrect club or a special race to get answered here. Terrible community.

khobar, 2020-05-11:
@billbop Github issues is not a forum , if you don't have anything related to that issue refrain yourself from posting.
Please refer to community guidelines https://help.github.com/en/github/site-policy/github-community-guidelines  ( especially point : Be clear and stay on topic ) 

Please do not reply

Joga29, 2020-05-12:
Ok guys. Rogue rg350 work on my Pocketgo V2 but i want to add opk's emulators , roms etc but the 2nd SD card IS not recognized do you have solution please ?

khobar, 2020-05-14:
Ensure that card is in correct format (for example ext4 and use ext2 volume manager to copy files ) mine second card was visible without any issues.
You can also try  via  network app ( there are some guides I think on net , have not tried this)

Also bitboy replied to my query ( which was before I created this issue on gh ) 

> The firmware for new pocketgo V2 upgraded version as same as the RG350 firmware.
You can install the RG350 firmware on your own sd card for a try. You can download the firmware from here: https://www.rghandhelds.com/
RetroGame RG-350 (Sept 2019)

TheIronCrab, 2020-05-25:
Hi. I have the same problem as @Joga29 . RG350 CFW is running well on mi New PocketGO V2, but 2nd SD Card is not recognized. I formated it to FAT32 with Windows 10. Did you found a solution? Thanks!

Ninoh-FOX, 2020-05-26:
Fixed in the version 1.0.8

closed 2020-05-26 by Ninoh-FOX