Ninoh-FOX/POCKETGO2_ROGUE_CFW #37: Problems with the mixer volume fix

SanguineBrah, 2020-05-10:
From GldRush98 on [Reddit](https://www.reddit.com/r/Bittboy/comments/gh0vki/pocket_go_2_first_revision_sound_fix/):
1. The Master control item doesn't show up in AlsaMixer until AFTER you have played some type of audio through the speaker. So it doesn't show up on boot.
3. The Master volume is not remembered after a reboot. You need to set the Master volume every time (only after first playing some sort of audio).

wermelinger, 2020-05-10:
I can confirm the same behavior with 1.0.7.
But nevertheless this is already a great achievement. The high volume was the biggest downside of the PG2.

GldRush98, 2020-05-11:
In addition to the above, the script that is supposed to be writing the volume level to the state file is not working properly. Not really sure why, but the file was empty. If I write a value in to it, change the Master level, then shutdown, the value saved in the state file does not get updated.

My hacky work around has been to add a startup script to play a silent wav file less than a second long, then set the Master volume level to the value I want. This seems to be working for now.  Great work on this so far though!!

GldRush98, 2020-05-11:
It looks like Master not showing up until a sound is played is a known issue according to this thread: https://github.com/Ninoh-FOX/POCKETGO2_ROGUE_CFW/issues/29

Shut3n, 2020-05-15:
Hi,

I replaced '/etc/init.d/S90volume.sh' with this 'S90alsaconf.sh' file:

```
#!/bin/sh

case "$1" in
	start)
		echo "alsa-config Starting.."
		/usr/sbin/alsactl restore -f /usr/local/etc/asound.state
		;;
	stop)
		echo "alsa-config Stopping.."
		/usr/sbin/alsactl store -f /usr/local/etc/asound.state
		;;
	*)
		exit 1
		;;

esac

exit 0
```

First time you boot, it's important to "do the open APP" (any sound application) for the Master volume to appear on Alsamixer. You can adjust volumes then to your comfort). To finish reboot the system, for the alsa settings to be stored on '/usr/local/etc/asound.state' (no problem because 'alsactl' command will manage to create it if not avaliable, and keep it updated from now).

If you're interested, my alsaconf.state finished like this (but not necessary to copy it, because it will be created automatically on first reboot) :

```
state.Intel {
	control.1 {
		iface MIXER
		name 'Channel Mode'
		value '2ch'
		comment {
			access 'read write'
			type ENUMERATED
			count 1
			item.0 '2ch'
			item.1 '4ch'
			item.2 '6ch'
		}
	}
	control.2 {
		iface MIXER
		name 'Front Playback Volume'
		value.0 31
		value.1 31
		comment {
			access 'read write'
			type INTEGER
			count 2
			range '0 - 31'
			dbmin -4650
			dbmax 0
			dbvalue.0 0
			dbvalue.1 0
		}
	}
	control.3 {
		iface MIXER
		name 'Front Playback Switch'
		value.0 true
		value.1 true
		comment {
			access 'read write'
			type BOOLEAN
			count 2
		}
[... 413 lines cut ...]
			access read
			type INTEGER
			count 2
			range '0 - 36'
		}
	}
	control.38 {
		iface MIXER
		name 'PCM Playback Volume'
		value.0 255
		value.1 255
		comment {
			access 'read write user'
			type INTEGER
			count 2
			range '0 - 255'
			tlv '0000000100000008ffffec1400000014'
			dbmin -5100
			dbmax 0
			dbvalue.0 0
			dbvalue.1 0
		}
	}
}
```

Ninoh-FOX, 2020-05-26:
Fixed in the version 1.0.8

closed 2020-05-26 by Ninoh-FOX