Ninoh-FOX/POCKETGO2_ROGUE_CFW #38: Tic-80 emulator

Tatohead, 2020-05-11:
I would say it's not anywhere close to top priority since it does works on an older cfw. I'm just reporting an issue that is seen on the recent CFW.

The Tic-80 emulator was working on the Rogue CFW 1.04. So far I only have tested CFW 1.05 and 1.07 both seem to have an issue with the emulator (I did not test 1.06 CFW yet). What occurred when I was attempting to run the roms/carts on the emulator, I got booted out. I'm not sure if this is just on my end.

I'm not sure if I'm providing any helpful information. I did looked at the log viewer to see if any messages showed up. Here's what it said:

lr: can't load library 'libshake.so.1'

I'm not familiar if there are rules, if putting link is not allowed I'll remove it.

The tic emulator opk in question.
https://drive.google.com/file/d/1OqrAW93IHF9eJwUwl1aDewHdEVUzfC3w/view?usp=sharing

Ninoh-FOX, 2020-05-11:
oh. I see, this is easy of fix, I am removed this library because only controlled the rumble, and this console doesn't has this. I am go to add this lib again in the next release then.

Ninoh-FOX, 2020-05-26:
fixed in the version 1.0.8

closed 2020-05-26 by Ninoh-FOX