Ninoh-FOX/POCKETGO2_ROGUE_CFW #39: Crackling sound

Arztiger, 2020-05-11:
So, after updating from 1.0.4 to 1.0.7. I wanted to play Castlevania: Rondo of Blood (in Temper emulator), to check the save files did they corrupt or no?  They did not, however some constant crackling sound appeared, like every 2-3 seconds. Experimenting with sound mixer did not help. This was only present in Temper, however, I tried doom 2 to be sure (on chocolate doom emulator) ...  crackling sound! I played with chocolate doom emulator settings, changed the sound encoding, and everything got back to normal. Kinda weird.

SanguineBrah, 2020-05-12:
I noticed some crackling when I set the master volume to 25, then it went away when I bumped it up to 50. Not sure if related.

Arztiger, 2020-05-12:
Well, I did not mention that while playing with sound mixer the crackling sound went away, but it comes back after restarting, and repeating the same "play" with sound mixer did not always fix the crackling, so the easy solution was just to roll back to the 1.0.4.

Retro-Ninja, 2020-05-13:
I've set mine back to its default setting and the sounds worked fine, i didn't need to downgrade.

Arztiger, 2020-05-14:
What settings are you talking about?

Shut3n, 2020-05-21:
Hi,

I leave my /etc/libao.conf like this:

```
default_driver=alsa
#buffer_time=80
#period_time=46440
#use_mmap=true
quiet
```

I tested on GenPlusgx and ReGBA, and it seems crackling dissapeared or it's less noticeable now.

Ninoh-FOX, 2020-05-24:
> Hi,
> 
> I leave my /etc/libao.conf like this:
> 
> ```
> default_driver=alsa
> #buffer_time=80
> #period_time=46440
> #use_mmap=true
> quiet
> ```
> 
> I tested on GenPlusgx and ReGBA, and it seems crackling dissapeared or it's less noticeable now.

I need test this

Arztiger, 2020-05-24:
> > Hi,
> > I leave my /etc/libao.conf like this:
> > ```
> > default_driver=alsa
> > #buffer_time=80
> > #period_time=46440
> > #use_mmap=true
> > quiet
> > ```
> > 
> > 
> > I tested on GenPlusgx and ReGBA, and it seems crackling dissapeared or it's less noticeable now.
> 
> I need test this

It's the same, however, 1 thing that I understood is that, after restarting the doom emulator or pc-engine Crackling goes away. I did not reply because it's all really confusing.

Arztiger, 2020-05-24:
And that seems to be random too, sometimes restarting works, sometimes not.

Ninoh-FOX, 2020-05-26:
yes, I located the problem, I am fixed this in the release 1.0.8

closed 2020-05-26 by Ninoh-FOX