Ninoh-FOX/POCKETGO2_ROGUE_CFW #40: Pocket Go V2 Buzzing Sound

Midas2310, 2020-05-21:
Hey i got my Pocket go V2 it runs atm the rg350 rouge fw but i got this problem with the screen and the buzzing sound. i found on guide from another user (a bit older) but this one dosent work with the new version. i wana ask if there is any way on the rg350 fw to increase the screen hz to stop the buzzing sound  ?

here is the old guide: https://byteporter.com/pocket-go-pwm-script-install/

hope someone can help me i cant get my self to use the pocket go like this becuase this sound makes me crazy its just to clear to here for me

Shut3n, 2020-05-21:
Hi,

Thank you for this article. I'm experiencing same buzzing on my Playgo handheld speaker. It starts when starting a rom from an emulator, and remains while on pause. It happens only for the speaker, not for the headphones. As I read on the article I tried increasing and lowing the screen bright, but nothing changed. About the scripts on the article, I couldn't find a way to use them because they are made for Retrofw system (PG1), and not the Opendingux that uses Pocketgo v2 or Playgo.

On other hand, a few days ago we were fixing this annoying loud volume level from the speaker, with some  adjustements in the ALSA parameters (for gcw0 audio drivers). Now alsamixer shows the Master volume control, which actually has enabled a virtual channel for it; but I think in a generic mode. So there's more improvements to be done, like some crackling sounds on some cases and this speaker buzzing.

Looking for more info on Alsa documentation (https://www.alsa-project.org/wiki/Main_Page), I found some audio cards drivers uses specific firmwares that improve hardware compatibility.

In this case by an specific component for jz4770-audio (gwc0), but I need for doing my tests these Alsa components to be installed on the system: alsa-firmware, aplay and alsaconfig (I couldn't find them on current -v107- rogue release).

Maybe including these firmwares, tools and components, I could manage to replace this "generic" audio channel for its native jz4770 codec, improving audio support  in more ways (and maybe output quality too).

Midas2310, 2020-05-21:
on me its the screen that makes the sound i alrdy tryed and removed the speakers and the sound was still there so i think is like in the guide from the other guy it the hz rate on the screen becuase he coud fix it by increasing the hz from 1k to 17k but i duno how to do this im not rly in to this stuff hope from help her.

myaple, 2020-05-25:
Correct me if I'm wrong @Ninoh-FOX, but this has already been modified: https://github.com/Ninoh-FOX/POCKETGO2_ROGUE_CFW/blob/master/kernel/arch/mips/jz4770/board-gcw0.c#L547

Ninoh-FOX, 2020-05-26:
> Correct me if I'm wrong @Ninoh-FOX, but this has already been modified: https://github.com/Ninoh-FOX/POCKETGO2_ROGUE_CFW/blob/master/kernel/arch/mips/jz4770/board-gcw0.c#L547

this is for the screen backlight time, anyway, the buzzing not is a problem of FW, is a problem of board, I ear the same sound if I tried power on the console without sdcard.

please, close the issues, the cfw don't can fixer this, but maybe an hardmod could.

closed 2020-07-02 by Ninoh-FOX