Ninoh-FOX/POCKETGO2_ROGUE_CFW #41: Lag spiking on V2 firmware 1.0.8

MrNB0ss, 2020-05-27:
So on my PocketGo V2, I have the 1.0.8 firmware installed. It lags at random times quite intensely, making games unplayable. This didn't do it on the stock firmware, so I'm wondering if this is a known issue?

GreatWhiteGT, 2020-05-27:
As another preface, I'm on the really new V2.1, which came out a couple months ago if I'm right.

Allexiel, 2020-05-27:
Hi! I can confirm.
I'm the lucky ... Owner of the 2, the 2v1 and the 2v2.
I have the same issue as described above. A clean install of 1.0.8 on 2v2 gives me lag and sound issues on SNES stock included emulator. Game tested was Super Mario World.
I have no lag spikes or sound issues on stock firmware ( have a spare sd card) nor on the 1.0.8 on the 2v1 Pocket Go

egeon44, 2020-05-28:
On PG2V1 with 1.0.8, even gameboy emulator have lag. Fba is unplayable

Ninoh-FOX, 2020-05-28:
> On PG2V1 with 1.0.8, even gameboy emulator have lag. Fba is unplayable

Do you have the emulators in 360mhz? Because works fine.

egeon44, 2020-05-28:
My mistake, after a fresh install on a new SDcard without redimensioning the ext4 partition, everything seems to work perfectly.

Ninoh-FOX, 2020-05-28:
> My mistake, after a fresh install on a new SDcard without redimensioning the ext4 partition, everything seems to work perfectly.

The base system and rogue not has the same partition file system, is for this that is necesary a clear install when you go to change of the CFW, before you only need update. I the readme can you read the steps.

Ninoh-FOX, 2020-05-28:
Anywere I go to update a news sdimages and update files for if anything was corrupter in the updater.

Cibomatto2002, 2020-05-31:
I have the same trouble but I have no idea what redimensioning the ext4 partition means. I use Etcher with the sd_image_PG2V2.bin
[image]

Here is a video of the problem https://www.youtube.com/watch?v=6rE72vrTdoA

Cibomatto2002, 2020-05-31:
I'm not sure what happen but now it's working correct i rebooted it and it was still not working right but I kept on playing it then over time the games got to where they stopped lagging and the sound got better.

GreatWhiteGT, 2020-06-05:
ReGBA and gbsp run terribly, apparently there's not alternatives either. Anyone know of better emulator opks?

Cibomatto2002, 2020-06-05:
ReGBA runs fine for me did you just install the 1.0.8 firmware?

famouswolfe, 2020-06-30:
Having issues as well with the Rogue CFW 1.0.8 firmware and GBA emulation. I even installed the GBA bios but Pokemon Fire Red looks like someone vomited up the opening text on my screen.

closed 2021-03-21 by Ninoh-FOX