Ninoh-FOX/POCKETGO2_ROGUE_CFW #46: Please publish image creation scripts

lubosz, 2020-06-14:
In your (buildroot fork)[https://github.com/Ninoh-FOX/toolchain] you have an apparently unused (make_initial_image.sh)[https://github.com/Ninoh-FOX/toolchain/blob/master/board/opendingux/pg2v2/make_initial_image.sh] for the PocketGo 2v2 with it accompanying (genimage.cfg)[https://github.com/Ninoh-FOX/toolchain/blob/master/board/opendingux/pg2v2/genimage.cfg].

I was able to manually copy a `mininit-syspart`, generate a `mininit-syspart.sha1` and execute the script after updating `genimage` to a less ancient version.

Although the resulting `sd_image.bin` differs significantly from the `sd_image_pg2.bin` you provide in your (release section)[https://github.com/Ninoh-FOX/POCKETGO2_ROGUE_CFW/releases].

The image is able to display the splash screen, but ends up in a resize / swap file creation boot loop and results in a corrupted app partition.
It also creates an extra hibernate partition which you don't provide in your `sd_image_pg2.bin` and also differs in size of the app partition.

Could you please publish your full build scripts, so others are able to reproduce an bootable image like `sd_image_pg2.bin` from source?

lubosz, 2020-06-14:
So after rebasing your git repository to gcwnow's and tonyjih's buildroot 2014.08 branches I noticed that you added an `imager` directory, which contains `GCW Zero Image Generation Tools port to RG350`. 

In [a8550f3aa7a5dbb476798f176987ac89d4a7834b](https://github.com/Ninoh-FOX/toolchain/commit/a8550f3aa7a5dbb476798f176987ac89d4a7834b) you moved that into the `Updaters` directory (which you later luckily renamed to lower case), where you keep one copy of the tool for every device. With very little changes between the directories.

So I apparently have 4 forks of the tool available:
No suffix, `m`, `pg` and `v2`. I assume `v2` is pocketgo2v2?
https://github.com/Ninoh-FOX/toolchain/tree/master/updaters/imager_v2

It would be really nice to have some documentation and less redundancy here.

Ninoh-FOX, 2020-07-02:
yes, v2 is pg2v2

Ninoh-FOX, 2021-02-16:
I am go to add a document soom

closed 2021-02-16 by Ninoh-FOX