Ninoh-FOX/POCKETGO2_ROGUE_CFW #47: [REQUEST] Autostart last played game

tuxfamily, 2020-07-01:
Hello.

It's probably not the right place for that but I would like to make a suggestion/request.
I think it would be useful (at least for me :p) to have an option to automaticaly start the last played game, at console start.
I don't know if it has already been requested, I googled that, and did not find anything relevant, so ... I ask :)

In the other hand, I tried to make a init script in /media/data/local/etc/init.d for that but it does not seems to be executed.
So, is there a quick/dirty way to have a "autostart" script, somewhere, until an official feature ? :D

Thank you.

davgonza, 2020-07-04:
Just wanted to comment on this one, and give a little visibility. +1 

I also agree this would be useful. You wouldn't even need a GUI for this necessarily, maybe just like a .config file or something in a shared directory, which contains the name of the ROM you want to load, after the boot screen appears on startup. But not if this config file doesn't exist, or there is nothing inside the config file. 

Of course, I understand if the way the code is structured, this might not be _technically_ possible.

Thanks

tuxfamily, 2020-08-03:
For the "dirty way", I found that it's possible to create a script nammed "frontend_start" inside the folder "/media/data/local/sbin" and that this script is executed automatically.

Here is my script:

```sh
#!/bin/sh
/usr/bin/opkrun "/media/data/apps/Regba.opk" /media/data/ROMS/gba/Legend\ of\ Zelda,\ The\ -\ The\ Minish\ Cap\ \(USA\).gba
# on exit -> we start gmenu2x
/usr/bin/gmenu2x
```
And it works pretty well.

I think it should be possible to develop a GUI to select the emulator and ROM to launch and then save these choice in this file. Just my 5 cents.

tuxfamily, 2020-08-10:
There is an issue with the above script: it well runs the game on the first launch, and gmenu2x on emulator exit, BUT, if you run another app and exit them, it will re-launch the autostarted game.

So to fix this issue, and launch the game automatically only once, on cold boot, I now create a temp file in order to check if the game has already be launched, or not. And as the tmp folder is cleaned at every start, it just works this way.

Here is the updated script:

```sh
#!/bin/sh
FILE=/tmp/launched
if [ -f "$FILE" ]; then
    /usr/bin/gmenu2x   
else                   
    touch "$FILE"   
    /usr/bin/opkrun "/media/data/apps/Regba.opk" /media/data/ROMS/gba/Legend\ of\ Zelda,\ The\ -\ The\ Minish\ Cap\ \(Patched\).gba
    /usr/bin/gmenu2x                                                                                                               
fi
```

Drumaster78, 2020-09-01:
Thanks for the tip, I just discovered the 'frontend_start' thing 2 days ago while looking at Esoteric launcher source code.

I was looking for a way to execute 'sync' command each time GMenu2X is called (when exiting an app for instance) to prevent my savestates from being wiped if the console is not turned off properly.

So now that works well this way. :)

Ninoh-FOX, 2021-03-29:
add recently that gmenu2x remembers the last rom released by emulator, I think it is a good alternative option to what you want