Ninoh-FOX/POCKETGO2_ROGUE_CFW #50: Sound crackling when save state

joelvdc, 2020-07-18:
This is a known issue, but I couldn’t see it raised here. I’m using rogue 1.0.9 and experiencing sound crackling when playing and using save states, and in some cases also when just saving games the regular way. I play mainly the ps1 and gba emulators, so I’m not sure if this issue is for all emulators. Restarting the emulator usually solves the issue, but it comes back as soon as I save state and it can get worst if I save state multiple times during a play session.

WLR86, 2020-08-04:
Same here as well

week0030, 2021-02-15:
Same ... and in ReGBA i my case when thats happend then FPS is not stable on 59/60 fps but then drops fluctuating from 20-60 fps. It looks like something highly consume cpu time and crackling is sound is consequence.

Ninoh-FOX, 2021-02-16:
try this opks compiled with https://github.com/jdgleaver/ codes

https://mega.nz/file/7hxV0KDQ#7Olif0hBiYI1e2MTFQ8bZLEej_ms5y832MnJ6-z4cJ0 last ReGBA for PG2/PG2v2

https://mega.nz/file/aow02RSY#Mn-PAPBkIV0sSVgpSAuPUj4L8QWz2k43CmFnnUf2n5A last PCSX4ALL for PG2

https://mega.nz/file/r8xGABYJ#xqVtCLKFqTHPwRHPUQOyC5wCxV8S38afdPII26s9MfY last PCSX4ALL for PG2v2

week0030, 2021-02-17:
> try this opks compiled with https://github.com/jdgleaver/ codes
>
> https://mega.nz/file/7hxV0KDQ#7Olif0hBiYI1e2MTFQ8bZLEej_ms5y832MnJ6-z4cJ0 last ReGBA for PG2/PG2v2 [...]

I'm dumb. I can't find issues on RG repo and then i came here from Google without checking right repo. I have this problem on RG350M, not on PG2, then i can't try mentioned builds. Can you build this for rg350m too?

Ninoh-FOX, 2021-02-17:
Well, then is how is I said that I have problems in psx games emulated (not playstore) in a psvita... 

You can download the RG emus version in the github of  
https://github.com/jdgleaver/

Ninoh-FOX, 2021-02-17:
Anyway. I am save and load states in rogue and I don't have problems.

No fps down, no sound crasher.

Have you done a clean installation of ROGUE?  because I have already said in other sites that updating from the stock image and even more so if the card that comes with the factory console is used, gives problems.

week0030, 2021-02-19:
I tried ReGBA from mentioned repo and i'am using same version. I'am not using original SD card. I tried two new quality class 10 cards. Same. Then i upgraded to 1.9.11u2 ( but it was worst .. it happened offen - i ussualy use 1.9.10). Then i made fresh install 1.9.10 again from bin and same. Sometimes i played hour a i made 20 saves and everything is ok but sometimes it happend after third save and that's happened if game is saved originaly from game via save point too! Wierd thing is i haven't this issue for long time. I played two Castelvania Games before without this issue. But now i played third Castelvania game ( Circle of Moon ) and i have it offten. I tried stock fw for few hours and nothing. 

About this issue is thread on reedit too: https://www.reddit.com/r/RG350/comments/ip58u9/heads_up_for_anyone_experience_crackling_issues/?utm_source=amp&utm_medium=&utm_content=post_title

There is advice to underclock ReGBA to 360 Mhz but this is bad advice, because after that most of games is unplayable. But it is true there is no issue anymore. But i try that only for 30 mins. Maybe overclock is true problem and my cpu is worst like yours. Its posible build fw without overclock ? Or implement same overclock options like in opendingux beta fw where is posible set up lot of frequencies ?

Sorry for my English and thx for yours interest.. and perfect fw of course :)

Ninoh-FOX, 2021-03-08:
> I tried ReGBA from mentioned repo and i'am using same version. I'am not using original SD card. I tried two new quality class 10 cards. Same. Then i upgraded to 1.9.11u2 ( but it was worst .. it [...]

yes, it is done

https://github.com/Ninoh-FOX/toolchain_cpu_stock/releases/tag/1.7.12-1.1

week0030, 2021-03-18:
> yes, it is done
>
> https://github.com/Ninoh-FOX/toolchain_cpu_stock/releases/tag/1.7.12-1.1

Nice. I was little bit busy. I updated now. After testing i send you info

Ninoh-FOX, 2021-03-18:
> Nice. I was little bit busy. I updated now. After testing i send you info

Use the update last update, please

week0030, 2021-03-19:
> > > > I tried ReGBA from mentioned repo and i'am using same version. I'am not using original SD card. I tried two new quality class 10 cards. Same. Then i upgraded to 1.9.11u2 ( but it was worst .. it happened offen - i ussualy use 1.9.10). Then i made fresh install 1.9.10 again from bin and same. Sometimes i played hour a i made 20 saves and everything is ok but sometimes it happend after third save and that's happened if game is saved originaly from game via save point too! Wierd thing is i haven't this issue for long time. I played two Castelvania Games before without this issue. But now i played third Castelvania game ( Circle of Moon ) and i have it offten. I tried stock fw for few hours and nothing.
> > > > About this issue is thread on reedit too: https://www.reddit.com/r/RG350/comments/ip58u9/heads_up_for_anyone_experience_crackling_issues/?utm_source=amp&utm_medium=&utm_content=post_title
> > > > There is advice to underclock ReGBA to 360 Mhz but this is bad advice, because after that most of games is unplayable. But it is true there is no issue anymore. But i try that only for 30 mins. Maybe overclock is true problem and my cpu is worst like yours. Its posible build fw without overclock ? Or implement same overclock options like in opendingux beta fw where is posible set up lot of frequencies ?
> > > > Sorry for my English and thx for yours interest.. and perfect fw of course :)
> > > 
> > > 
> > > yes, it is done
> > > https://github.com/Ninoh-FOX/toolchain_cpu_stock/releases/tag/1.7.12-1.1
> > 
> > 
> > Nice. I was little bit busy. I updated now. After testing i send you info
> 
> Use the last update, please

Hi version 1.7.12-1.1 dosen't help. Today i tried newer version like you say. BUT now i have error in log if happened from Psx4All:

ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occured

[image]

Ninoh-FOX, 2021-03-19:
> Hi version 1.7.12-1.1 dosen't help. Today i tried newer version like you say. BUT now i have error in log if happened from Psx4All:
>
> ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occured
>
> [image]

The last update is 1.3 o 1.7.14-1.3

closed 2021-03-21 by Ninoh-FOX

week0030, 2021-03-21:
Yes with 1.7.14 it is fixed! I tried most of emus and no problem anymore. Thx!