Ninoh-FOX/POCKETGO2_ROGUE_CFW #58: Help something is wrong after updating on playgo

adi170-alt, 2021-03-09:
https://imgur.com/a/b9ZaOZ1

HELP what do I need to do now???
after that message it reboots and shows that again

and when I press X and Y at boot it shows the boot screen before the update and after that a black screen
I have proof that it just works fine with that software but I really want to keep my files
My motherboard number is: GCWZERO_V1.2 2019.11.26
and I used just the PlayGo-update-1.0.9-2020-07-02.opk file to update it

_Originally posted by @adi170-alt in https://github.com/Ninoh-FOX/POCKETGO2_ROGUE_CFW/issues/57#issuecomment-793646251_

Ninoh-FOX, 2021-03-10:
that seems to be a microsd problem, do you use the factory microsd?

Download this PlayGo-update-1.1-2021-03-01.opk and unzip the file with 7zip in the pc, take the files mininit-syspart, modules.squashfs, rootfs.squashfs and vmlinuz.bin and put this in the root of the fat32 partition of the microsd (first partition) and erase the other files, not erase the folder.

Then if you use linux, open gparted and check the ext4 partition of the microsd (second partition), if you don't have linux, you can't use a linux imagen livecd in an usb pendrive for boot this in the pc and run gparted.

With this, the console should boot fine. But if you use the microsd of stock console, is better changer this for another, for example sandisk ultra 16gb.

adi170-alt, 2021-03-10:
Hi, it boots up now but I have a problem again:( , it boots ups now just fine but I can not control it, the only button I can use is the power button, and with that i can only set it to sleep mode or completely off, And yes I use the factory micro sd card but I'm gonna change it, I just care about the games, saves, emulators and roms for now, and I can not get those files just by usb because there is a password and I need to turn that off but i can not control it

Ninoh-FOX, 2021-03-10:
The only option is copy the home, data and app folder from pc

adi170-alt, 2021-03-10:
how??, when i put the micro sd in my pc ten there is no home directory, please tell me how, and when I connect to my pc,  then I can go ftp but i dont know the pasword, and the pasword you only can get by controll it to a app

adi170-alt, 2021-03-10:
oke, oke. tnx