Ninoh-FOX/POCKETGO2_ROGUE_CFW #61: Stock Clock Rogue PG2 using save states locks up unit

soulsonic57, 2021-03-17:
Hi Ninoh and community,

I installed the latest rogue cfw on my PG2 and unfortunately excessive noises of the components cooling down led me to be concerned that the clockspeed may reduce the lifespan of my unit. I therefore decided to flash a new card with the stock clock rogue cfw.

Unfortunately, I have encountered an issue whereby if I use save states on a game running on an emulator, when I try to run a different rom using the same emulator I get a black screen rendering the unit ‘locked up’. I obviously have to perform a hard reset in order to get the unit to be usable again. However, if I choose to run the same rom which had the save state, the unit locks up again necessitating another hard reset.

Can you advise please? Do you intend to patch this issue. I am running the base .bin with the update .opk from 13/3/21 on a San Disk class 10 32Gb card.

Thanks in advance.

Best regards,
Soulsonic57

Ninoh-FOX, 2021-03-21:
But WHAT emulator?

soulsonic57, 2021-03-25:
Apologies for my belated reply Ninoh. The emulators in question are Genesis Plus GX and PocketSNES. I can confirm that on both emulators, the problem is not limited to a single rom file. I hope this helps. If you need further info, please let me know.