Ninoh-FOX/POCKETGO2_ROGUE_CFW #63: Error while updating....

MaxS78787, 2021-03-22:
Hey there, 

I'm not sure what I'm doing wrong but I get that error when I update with the opk on Pocket Go v2 
"Updated RootFS is corrupted!"

Everything goes fine for a while but at the end when it checks it says this - and the update doesn't seem to have happened on reboot.. 

Any ideas? Thanks!

[image]

Ninoh-FOX, 2021-03-22:
that error usually appears sometimes when the sha1 does not match or some file has not been copied well. Although I still don't know why it comes out to some people in particular.

You can always update manually in two ways unzipping the update opk with 7zip and removing the files modules.squashfs, modules.squashfs.sha1, rootfs.squashfs, rootfs.squashfs.sha1, vmlinuz.bin and vmlinuz.bin.sha1

way ftp:

configure filezilla ftp sever as SFTP - SSH file transfel protocol and drag the files that I told you before to the folder:

/ media / system

and overwrite the files, then from terminal / cmd you write:

- ssh root@10.1.1.2
- reboot

way pc:

put the microsd in the computer, and in the partition fat32 of 400Mg put and overwrite the files that I have said before.

MaxS78787, 2021-03-22:
Thank you for the fast response! Updating manually like you said seem to have work perfectly

(I also added the mininit-syspart.sha1, and mininit-syspart - idk if you forgot them or if they were not necessary but I saw them in the folder to overwrite) 

Thank you so much for your work, it is much appreciated!

closed 2021-03-22 by Ninoh-FOX