Ninoh-FOX/POCKETGO2_ROGUE_CFW #64: fba (cps1/2) games don't work

adi170-alt, 2021-03-22:
[https://imgur.com/a/U7NBiEP](url)
I tried to play a few cps 1 and 2 games but every way I try to play them that in the picture pops up or when I click nothing happens,
I tried to use XMAME, but that also did not work because the games did not show(I putted the games in the rom's folder). It is hard to explain why, but I don't use the fba emulator that came when you flashed it on a sd-card.

Ninoh-FOX, 2021-03-22:
Are you really sure you don't have the right romset before blaming the firmware?

[image]

Ninoh-FOX, 2021-03-22:
Are you really sure you have the roms in the right place before blaming the firmware?
[image]

Ninoh-FOX, 2021-03-22:
you problem, not is the firmware/emulator, is that your roms versions **not is the correct** for the emulators.

closed 2021-03-22 by Ninoh-FOX

adi170-alt, 2021-03-22:
> Are you really sure you don't have the right romset before blaming the firmware?
> 
> [image]

I don't blame the firmware, it is just the only place that I can come to and ask questions, because I'm scarred that I am going to get hate or almost no attention on Reddit.

Ninoh-FOX, 2021-03-22:
as I have already told you, you need the correct romset of roms, I can not give you support in that

adi170-alt, 2021-03-23:
I know