Ninoh-FOX/POCKETGO2_ROGUE_CFW #68: Console reboots after some usage with latest release

squidjam, 2021-09-12:
I'm using the latest release and while playing mame4all and rrootage I noticed the colors suddenly shifting to one side and the OS reloads itself showing the "Rogue" loading screen.

I tried switching to the original SD card and had no issues.

Device: PlayGo.
Battery level: 91%
Release:  v1.4.2
Installed or Updated: Installed from scratch into  32GB SanDisk new SD card.

Ninoh-FOX, 2021-09-12:
Do you have this problem too in version 1.4?

Ninoh-FOX, 2021-09-19:
I have tried with xmame for hours and I have not seen problems in my v1, very few consoles that affect the overclock, maybe it is the problem, what you can do is the following: 
1. copy the files from the fat32 partition to the pc .  
2. update the kernel to the stock clock version.  
3. Replace the files that you now have in the fat32 partition with the ones you copied to the pc.  
 
This way you will have version 1.4.2 without overclock.

closed 2021-12-18 by Ninoh-FOX